Marzogh/SPIMemory #202: Cannot erase memory partially with Winbond W25Q16BV

sabuto26, 2020-03-09:
##### . Every function i use erase all the memory(****eraseBlock32**(addr), **eraseSector**(), **eraseSection**()** ). But i want to erase just one address.

Thank You!

##

sabuto26, 2020-03-09:
`String sc = "678000000";
    String sc1 = "DCDBBBBV";
    //Serial.println(arrayLen(wr_dt));
    //for (uint8_t i = 0; i < arrayLen(wr_dt); i++) 
    {
      if (pin6_flash.eraseBlock32K(2096992)) 
      {
        Serial.println("Sector has been erased");
        delay(3000);
        if (pin6_flash.writeStr(2096992, /*wr_dt[0]*/sc)) 
        {
          Serial.print("'");
          Serial.print(wr_dt[0]);
          Serial.print("' written to :");
          Serial.println(2096992);
          delay(500);
        }
        delay(500);
      }
      if (pin6_flash.eraseBlock32K(2096944)) 
      {
        Serial.println("Sector has been erased");
        delay(3000);
        if (pin6_flash.writeStr(2096944, sc1/*wr_dt[1]*/)) 
        {
          Serial.print("'");
          Serial.print(wr_dt[1]);
          Serial.print("' written to :");
          Serial.println(2096944);
          delay(500);
        }
      }`

sabuto26, 2020-03-10:
Sorry !

Now, i understand the functions.

> Takes an address as the argument and erases the block containing the 

> address.

Thank!

03sarath, 2020-03-26:
@sabuto26 ,I hope you doing good,

Do you get any solution for erasing **one address**?I'm facing the same problem

intensite, 2020-03-26:
I think for erasing one full block at a time is a limitation/design of Flash Memory in general.
According to this: https://computer.howstuffworks.com/flash-memory1.htm

> The electrons in the cells of a flash-memory chip can be returned to normal ("1") by the application of an electric field, a higher-voltage charge. Flash memory uses in-circuit wiring to apply the electric field either to the entire chip or to predetermined sections known as blocks. This erases the targeted area of the chip, which can then be rewritten. Flash memory works much faster than traditional EEPROMs because instead of erasing one byte at a time, it erases a block or the entire chip, and then rewrites it.

03sarath, 2020-03-27:
@intensite well said thank you.