Marzogh/SPIMemory #206: Bug report - STM32F103RCT6 run diagnostics failed

santhosh1211, 2020-04-10:
##### Any bug report raised here MUST be submitted according to this template or it will be flagged with 'Not enough information'. No action will be taken till all the prerequisite information is provided. If no information is provided for over a month after the 'Not enough information' label is applied, the issue will be closed.

## Pre-bug report checklist:

Do this checklist before filing an bug report:
- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post it as a Feature request. NOT a bug report.
- [ ] Is this a bug that you cannot fix? Go ahead with filing a bug report below

<hr>

## Bug Report
HI Marzogh,

i am using stm32f103rct6 controller i connected default spi pins to the W25Q64  SPI flash .

i but the ARCH_STM32 line in the variant.h file after that its complied.

I tried multiple SPI clock speed setting after that Erase Chip and Power Up	function start working 

but the run diagnostic is getting failed . so that read write function also not working .

Kindly support me to solve this issue . 

#### Describe the bug

_Include a clear and concise description of what the bug is._

- Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. Paste the **relevant sections** of the output from your Serial console when you run FlashDiagnostics.ino.here:
```
Initialising..........

SPIMemory Library version: < 2.5.0

JEDEC ID: 0x77A00B
Man ID: 0x77
Memory ID: 0xA0
Capacity: 0
Max Pages: 0
Unique ID: 2241195777991936, 0x6458811BE382500
-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down		   PASS				      0 us
			Power Up		   PASS				      0 us

			Erase Chip		   PASS				      0 us
			Erase 72KB		   FAIL	
			Erase 64KB		   FAIL	
			Erase 32KB		   FAIL	
			Erase 4KB		   FAIL	
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   FAIL				      0 us
			Char			   FAIL				      0 us
			Word			   FAIL				      0 us
			Short			   FAIL				      0 us
			ULong			   FAIL				      0 us
			Long			   FAIL				      0 us
			Float			   FAIL				      0 us
			Struct			   FAIL				      0 us
			Byte Array		   FAIL	1, 	   FAIL	2, 	   FAIL	3, 	   FAIL	4, 	   FAIL	5, 	   FAIL	6, 	   FAIL	7, 	   FAIL	8, 	   FAIL	9, 	   FAIL	10, 	   FAIL	11, 	   FAIL	12, 	   FAIL	13, 	   FAIL	14, 	   FAIL	15, 	   FAIL	16, 	   FAIL	17, 	   FAIL	18, 	   FAIL	19, 	   FAIL	20, 	   FAIL	21, 	   FAIL	22, 	   FAIL	23, 	   FAIL	24, 	   FAIL	25, 	   FAIL	26, 	   FAIL	27, 	   FAIL	28, 	   FAIL	29, 	   FAIL	30, 	   FAIL	31, 	   FAIL	32, 	   FAIL	33, 	   FAIL	34, 	   FAIL	35, 	   FAIL	36, 	   FAIL	37, 	   FAIL	38, 	   FAIL	39, 	   FAIL	40, 	   FAIL	41, 	   FAIL	42, 	   FAIL	43, 	   FAIL	44, 	   FAIL	45, 	   FAIL	46, 	   FAIL	47, 	   FAIL	48, 	   FAIL	49, 	   FAIL	50, 	   FAIL	51, 	   FAIL	52, 	   FAIL	53, 	   FAIL	54, 	   FAIL	55, 	   FAIL	56, 	   FAIL	57, 	   FAIL	58, 	   FAIL	59, 	   FAIL	60, 	   FAIL	61, 	   FAIL	62, 	   FAIL	63, 	   FAIL	64, 	   FAIL	65, 	   FAIL	66, 	   FAIL	67, 	   FAIL	68, 	   FAIL	69, 	   FAIL	70, 	   FAIL	71, 	   FAIL	72, 	   FAIL	73, 	   FAIL	74, 	   FAIL	75, 	   FAIL	76, 	   FAIL	77, 	   FAIL	78, 	   FAIL	79, 	   FAIL	80, 	   FAIL	81, 	   FAIL	82, 	   FAIL	83, 	   FAIL	84, 	   FAIL	85, 	   FAIL	86, 	   FAIL	87, 	   FAIL	88, 	   FAIL	89, 	   FAIL	90, 	   FAIL	91, 	   FAIL	92, 	   FAIL	93, 	   FAIL	94, 	   FAIL	95, 	   FAIL	96, 	   FAIL	97, 	   FAIL	98, 	   FAIL	99, 	   FAIL	100, 	   FAIL	101, 	   FAIL	102, 	   FAIL	103, 	   FAIL	104, 	   FAIL	105, 	   FAIL	106, 	   FAIL	107, 	   FAIL	108, 	   FAIL	109, 	   FAIL	110, 	   FAIL	111, 	   FAIL	112, 	   FAIL	113, 	   FAIL	114, 	   FAIL	115, 	   FAIL	116, 	   FAIL	117, 	   FAIL	118, 	   FAIL	119, 	   FAIL	120, 	   FAIL	121, 	   FAIL	122, 	   FAIL	123, 	   FAIL	124, 	   FAIL	125, 	   FAIL	126, 	   FAIL	127, 	   FAIL	128, 	   FAIL	129, 	   FAIL	130, 	   FAIL	131, 	   FAIL	132, 	   FAIL	133, 	   FAIL	134, 	   FAIL	135, 	   FAIL	136, 	   FAIL	137, 	   FAIL	138, 	   FAIL	139, 	   FAIL	140, 	   FAIL	141, 	   FAIL	142, 	   FAIL	143, 	   FAIL	144, 	   FAIL	145, 	   FAIL	146, 	   FAIL	147, 	   FAIL	148, 	   FAIL	149, 	   FAIL	150, 	   FAIL	151, 	   FAIL	152, 	   FAIL	153, 	   FAIL	154, 	   FAIL	155, 	   FAIL	156, 	   FAIL	157, 	   FAIL	158, 	   FAIL	159, 	   FAIL	160, 	   FAIL	161, 	   FAIL	162, 	   FAIL	163, 	   FAIL	164, 	   FAIL	165, 	   FAIL	166, 	   FAIL	167, 	   FAIL	168, 	   FAIL	169, 	   FAIL	170, 	   FAIL	171, 	   FAIL	172, 	   FAIL	173, 	   FAIL	174, 	   FAIL	175, 	   FAIL	176, 	   FAIL	177, 	   FAIL	178, 	   FAIL	179, 	   FAIL	180, 	   FAIL	181, 	   FAIL	182, 	   FAIL	183, 	   FAIL	184, 	   FAIL	185, 	   FAIL	186, 	   FAIL	187, 	   FAIL	188, 	   FAIL	189, 	   FAIL	190, 	   FAIL	191, 	   FAIL	192, 	   FAIL	193, 	   FAIL	194, 	   FAIL	195, 	   FAIL	196, 	   FAIL	197, 	   FAIL	198, 	   FAIL	199, 	   FAIL	200, 	   FAIL	201, 	   FAIL	202, 	   FAIL	203, 	   FAIL	204, 	   FAIL	205, 	   FAIL	206, 	   FAIL	207, 	   FAIL	208, 	   FAIL	209, 	   FAIL	210, 	   FAIL	211, 	   FAIL	212, 	   FAIL	213, 	   FAIL	214, 	   FAIL	215, 	   FAIL	216, 	   FAIL	217, 	   FAIL	218, 	   FAIL	219, 	   FAIL	220, 	   FAIL	221, 	   FAIL	222, 	   FAIL	223, 	   FAIL	224, 	   FAIL	225, 	   FAIL	226, 	   FAIL	227, 	   FAIL	228, 	   FAIL	229, 	   FAIL	230, 	   FAIL	231, 	   FAIL	232, 	   FAIL	233, 	   FAIL	234, 	   FAIL	235, 	   FAIL	236, 	   FAIL	237, 	   FAIL	238, 	   FAIL	239, 	   FAIL	240, 	   FAIL	241, 	   FAIL	242, 	   FAIL	243, 	   FAIL	244, 	   FAIL	245, 	   FAIL	246, 	   FAIL	247, 	   FAIL	248, 	   FAIL	249, 	   FAIL	250, 	   FAIL	251, 	   FAIL	252, 	   FAIL	253, 	   FAIL	254, 	   FAIL	255, 	   PASS				      0 us
			String			   FAIL				      0 us
-----------------------------------------------------------------------------------------------------------------------------
To see function runtimes ncomment RUNDIAGNOSTIC in SPIMemory.h.

```
[... 96 lines cut ...]
If applicable, add a wiring diagram to help explain your problem.

#### Software environment (please complete the following information):
 - Library version: [e.g. v3.4.0]
 - Arduino IDE version: [e.g. 1.8.11]
 - OS: [e.g. windows]
 - Micro controller platform: [e.g.STM32F103RCT6]
 - Flash/Fram memory module: [e.g.25Q64BVSIG] WINBOND

#### Additional context
Add any other context about the problem here.

<hr>

###### DO NOT DELETE OR EDIT anything below this

<hr>

<sub> <b> Note 1: _Make sure to add **all the information needed to understand the bug** so that someone can help. If any essential information is missing we'll add the 'Needs more information' label and close the issue until there is enough information._ </b></sub>

<sub> <b> Note 2: For support questions (for example, tutorials on how to use the library), please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports. </b></sub>

<hr>

[image] [image] [image] [image] [image] [image] [image]

santhosh1211, 2020-04-10:
HI Marzogh,

i changed the default clock speed to 18mhz now its working fine ,

flash.setClock(2000000);  this function is working but not 100% after changing the below modification its working fine for STM32f103

Modification found from (Bug report - Not working by default with Teensy 4.0 and W25Q64JV #199 )
Whereas the SPIMemory library sets this to 104MHz by default (in defines.h):

#define SPI_CLK 104000000 //Hz equivalent of 104MHz

Changing SPI_CLK to "#define SPI_CLK 18000000" 

its working good .

Thank you

delgadosouza, 2020-05-24:
Thank you santhosh1211.
That was what I needed for it to work flawlessly, reducing the speed of the SPI CLK.
I am using W25Q128FVSG module.
My board is the BluePill STM32F103C8T6.

I noticed that the ARCH_STM32 is not selected automatically so the constructor was trying to match other patterns. I don't know if I should have written #define ARCH_STM32 at the beggining of the sketch though. For a final try I reinstalled the library to get rid of any modifications and I put this line at the beggining and it now selects the right constructor.

closed 2020-05-30 by santhosh1211