Marzogh/SPIMemory #221: W25Q128JV fails with error code 2, "Unable to identify chip"

MaffooClock, 2021-04-06:
##### Any bug report raised here MUST be submitted according to this template or it will be flagged with 'Not enough information'. No action will be taken till all the prerequisite information is provided. If no information is provided for over a month after the 'Not enough information' label is applied, the issue will be closed.

## Pre-bug report checklist:

Do this checklist before filing an bug report:
- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post it as a Feature request. NOT a bug report.
- [x] Is this a bug that you cannot fix? Go ahead with filing a bug report below

<hr>

## Bug Report

#### Describe the bug

I'm not certain this is a bug, but everything I try yields the exact same result.  I've got a [SparkFun Serial Flash Breakout](https://www.sparkfun.com/products/17115), which employs a 16Mbyte [Winbond W25Q128JV](https://www.winbond.com/resource-files/W25Q128JV%20RevH%2003102021%20Plus.pdf), connected to an [Arduino MKR GSM1400](https://store.arduino.cc/usa/mkr-gsm-1400).

Initialization with `flash.begin()` results in an `0x02` error, "Unable to identify chip. Are you sure this chip is supported?"

I have several of these breakout boards, and I've tried two of them, same results.

I have verified that 3.3v is present, and that GND and all SPI lines have continuity.

Normally, I would suspect reversed MISO/MOSI, but that is not the case (I've checked several times to be sure), or perhaps a bad chip (but I've tried two so far).  I know the chip is communicating because I can read the JEDECID.

If I'm being objective, I'd say that my issue appears to be a hardware or other user fault and not a bug in the library.  The only reason, then, to file the bug is based on the README, which states "Should work with the ... W25QXXXJV families," but it hasn't specifically been tested.

- Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. Paste the **relevant sections** of the output from your Serial console when you run FlashDiagnostics.ino.here:
```
Initialising..........
Chip Diagnostics initiated.

Unable to identify chip. Are you sure this chip is supported?

SPIMemory Library version: < 2.5.0

JEDEC ID: 0xF7380C
Man ID: 0xF7
Memory ID: 0x38
Capacity: 0
Max Pages: 0
Unique ID: 3743781493545216, 0x34163933E0A1900
---------------------------------------------------------------------------------------------------------
							Testing library code
---------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
---------------------------------------------------------------------------------------------------------
			Power Down		   PASS				      53 us
			Power Up		   FAIL	

			Erase Chip	The chip is currently powered down.
	   FAIL	
			Erase 72KB	The chip is currently powered down.
	   FAIL	
			Erase 64KB	The chip is currently powered down.
	   FAIL	
			Erase 32KB	The chip is currently powered down.
	   FAIL	
			Erase 4KB	The chip is currently powered down.
	   FAIL	
---------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
---------------------------------------------------------------------------------------------------------
The chip is currently powered down.
The chip is currently powered down.
			Byte			   FAIL			0 us			1.799 s
The chip is currently powered down.
The chip is currently powered down.
			Char			   FAIL			0 us			1.840 s
The chip is currently powered down.
The chip is currently powered down.
			Word			   FAIL			0 us			1.881 s
The chip is currently powered down.
The chip is currently powered down.
			Short			   FAIL			0 us			1.922 s
The chip is currently powered down.
The chip is currently powered down.
			ULong			   FAIL			0 us			1.963 s
The chip is currently powered down.
The chip is currently powered down.
			Long			   FAIL			0 us			2.004 s
The chip is currently powered down.
The chip is currently powered down.
			Float			   FAIL			0 us			2.045 s
The chip is currently powered down.
The chip is currently powered down.
			Struct			   FAIL			0 us			2.086 s
The chip is currently powered down.
The chip is currently powered down.
			Byte Array		   (trimmed this section -- all fail)
The chip is currently powered down.
The chip is currently powered down.
			String			   FAIL			0 us			2.219 s
---------------------------------------------------------------------------------------------------------

```

#### To Reproduce
Provide a **minimal code snippet** example that reproduces the bug (If you're using one of the examples that came with the library, just tell us which one instead of pasting the entire example's code). Please make sure you wrap any code in the proper code blocks like below
```
#include <Arduino.h>
#include <SPIMemory.h>

#define CS_PIN 7

SPIFlash flash( CS_PIN );

void setup() {

  // Seems to fail at SPIFlash::_chipID() in SPIFlashIO.cpp
  // (the loop on _supportedManID doesn't have 0xF7?)

  if( !flash.begin() ) {
    Serial.println(F( "Error ));
    flash.error( VERBOSE );
  }
}

void loop() {
}
```

#### Expected behavior
I expect `flash.begin()` to return `true`.

#### Screenshots
N/A

#### Wiring diagram

```plain
W25Q128JV  ->  MKR GSM 1400
     3.3V  ->  VCC
     CS    ->  D7
     CLK   ->  SCLK
     DI    ->  MOSI
     DO    ->  MISO
     WP    ->  VCC
     HLD   ->  VCC
     GND   ->  GND
```

#### Software environment (please complete the following information):
 - Library version: v3.4.0 (output above says `< 2.5.0`, which is strange -- I definitely have 3.4.0)
 - Arduino IDE version: v1.8.11
 - OS: MacOS
 - Micro controller platform: Arduino
 - Flash/Fram memory module: W25Q128JV

#### Additional context
N/A

<hr>

###### DO NOT DELETE OR EDIT anything below this

<hr>

<sub> <b> Note 1: _Make sure to add **all the information needed to understand the bug** so that someone can help. If any essential information is missing we'll add the 'Needs more information' label and close the issue until there is enough information._ </b></sub>

<sub> <b> Note 2: For support questions (for example, tutorials on how to use the library), please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports. </b></sub>

<hr>

[image] [image] [image] [image] [image] [image] [image]

MaffooClock, 2021-04-06:
Update: I just realized that the manufacturer ID of the W25Q128JV is 0xF7, but Winbond is 0xEF in this library.  Regardless, setting a `#define CAPACITY MB(16)` and `flash.begin( CAPACITY );` did not work.

I switched to the stable branch, which seems to change a lot about how the chip ID is handled (Commit 5bb2267f7f090d94389b57a68c8b3e96f7d1dd26).  The W25Q128JV still isn't supported directly, but at least now defining a capacity makes it work.

Not sure if actually reading/writing will work...

tickietackie, 2021-04-16:
I have a w25q64 also with manufacture id 0xF7. I am struggling with the same issues. Did you get it to work now? I tried to define the capacity as you said, but I cannot get the diagnostics to run. You said you used a different branch. Can I just downgrade via the library manager to it? I am using an Arduino NANO 33 IoT.

tickietackie, 2021-04-16:
Ah yeah I see what you meant. With version 3.2.0 I get the same output as you. The capacity is not 0 then anymore, but still the power up is failing.

PaulZC, 2021-04-28:
When using the [SparkX board](https://www.sparkfun.com/products/17115), you do need to pull up the WP/2 and HLD/3 pins (if you are using standard SPI) otherwise it will fail to communicate.
Best wishes,
Paul

MaffooClock, 2021-04-28:
> When using the [SparkX board](https://www.sparkfun.com/products/17115), you do need to pull up the WP/2 and HLD/3 pins (if you are using standard SPI) otherwise it will fail to communicate.

Duh.

bmattedi, 2021-05-05:
Did it work?

I'm currently using the W25Q128JV, with 128Mb, and it works...

```
Initialising..........
Chip Diagnostics initiated.

No Chip size defined by user. Checking library support.
Chip identified. This chip is fully supported by the library.

SPIMemory Library version: < 2.5.0

JEDEC ID: 0xEF7018
Man ID: 0xEF
Memory ID: 0x70
Capacity: 16777216
Max Pages: 65536
Unique ID: 4093014330255360, 0x6014735329372800
-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down		   PASS				      31 us
			Power Up		   PASS				      24 us

			Erase Chip		   PASS				      41.830 s
			Erase 72KB		   PASS				      293.14 ms
			Erase 64KB		   PASS				      260.02 ms
			Erase 32KB		   PASS				      170.27 ms
			Erase 4KB		   PASS				      45.28 ms
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   PASS			72 us			24 us
			Char			   PASS			72 us			24 us
			Word			   PASS			76 us			28 us
			Short			   PASS			76 us			28 us
			ULong			   PASS			88 us			28 us
			Long			   PASS			84 us			28 us
			Float			   PASS			88 us			28 us
			Struct			   PASS			200 us			56 us
			Byte Array		   PASS			1.56 ms			368 us
			String			   PASS			296 us			100 us
-----------------------------------------------------------------------------------------------------------------------------

```

Simba5LK, 2022-10-19:
Hello, i'm using the library to access the W25Q16jv chip which is inbuilt in MKRWAN 1310. i have run the example program from the library. the flash.begin () is returning false and i also cannot read the JEDEC ID . I did uncommented the line #define RUNDIAGNOSTICS in the .h file.
I also tried the flash.error() , it is returning to 0x02. the chip is not getting identified. 
how can i correct this error?